boschresearch/blech #1: Where can I find documentation on the language?

frameworklabs, 2020-01-11:
Please provide or link to a file listing and explaining the available language constructs.

schorg, 2020-01-11:
Examples and a user manual will follow in the next weeks.
Stay tuned!

schorg, 2020-03-25:
A can find a first draft of the  [Blech user manual](https://github.com/boschresearch/blech-doc/blob/master/documentation/blech-user-manual.adoc) in the blech-doc repository [Blech documentation and website repository](https://github.com/boschresearch/blech-doc)
From this repository we will soon create documentation and a website.

FriedrichGretz, 2020-04-30:
The [website](https://boschresearch.github.io/blech-doc/) is now live and the aforementioned user manual can be found [there](https://boschresearch.github.io/blech-doc/docs/user-manual/).
Of course, it never is 100% complete. But I guess we can close the issue of having no doc at all. Further discussions concerning the details of the doc can be started on the issue tracker of the [documentation repository](https://github.com/boschresearch/blech-doc/issues).

closed 2020-04-30 by FriedrichGretz