boschresearch/blech #4: error in compiled C code

mterber, 2020-01-12:
The following activity code compiles in Blech without any errors:
```
activity SendCommand (spiReady: bool, cmd: uint8, data: [RF_MAX_CMD_DATA]uint8, len: uint8)
    // ord( ((cmd<<5)|addr), &data_byte, 1);
    var txData: SpiBuffer
    txData.mem[0] = cmd
    txData.len = 1
    param size: uint8 = len + 1
    // while txData.len < size repeat
    //     txData.mem[txData.len] = data[txData.len-1]
    //     txData.len = txData.len + 1
    // end

    _ = run SpiTxRxBytes(spiReady, txData)
end
```

The resulting C code, however, is rejected by the C compiler:
```
../Core/Blech/blech/control.c:112:42: error: 'blc_len' undeclared here (not in a function); did you mean 'blc_pc_t'?
 static blc_uint8 blc_SendCommand_size = (blc_len + 1);
                                          ^~~~~~~
                                          blc_pc_t
```

schorg, 2020-01-13:
This is a combination of errors in the analyis phases.
For the time being you should declare size as read-only
```
let size: uint8 = len + 1
```

FriedrichGretz, 2020-01-13:
I've tried to reproduce your issue with a minimal example using only the involved data elements:
```
@[EntryPoint]
activity SendCommand (len: uint8)
    
    param size: uint8 = len + 1
    
    await false
end
```
However the compiler correctly detects 
error: The static paramter size was initialised by SendCommand.len + 1 which assumes a value at runtime. Instead it must be initialised using only constants or other static paramters.
 --> .\sendCommand.blc:4:11 [typing]

Please check if your build of blechc is up to date.

mterber, 2020-01-14:
I can confirm that this issue was caused by an outdated version of the Blech compiler. I actually compiled and updated the compiler executable correctly but my helper script, which executes the Blech compiler for every *.blc file in my project, still pointed to an older version of the Blech compiler. The same is true for my VS code plugins which where also based on the older compiler version.

So from my point of view this issue is fixed.

closed 2020-01-15 by FriedrichGretz