boschresearch/blech #7: support for hex literals

mterber, 2020-01-29:
**Is your feature request related to a problem? Please describe.**
Not only but especially in the embedded context, it is common to write integer literals as hex values. This is particularly true for hardware manuals that define register addresses and commands in order to interact with the device. Today, writing hex literals is not supported in Blech. Thus, the user has to "translate" every value, e.g. from a microcontroller manual, into decimal values. This is a error-prone task and makes it difficult to cross-check source code and hardware documentation.

**Describe the solution you'd like**
I would like to be able to write something like `var cmd: uint8 = 0xC3` or `var data: [3]uint8 = {0xA4, 0x05, 0xF7}` in Blech.

schorg, 2020-01-30:
Startet to implemented this in feature branch 'feature/bits-and-bytes'.
There is an evolution proposal how to do this:  [bits-and-bytes.adoc](https://github.com/boschresearch/blech-doc/blob/master/documentation/proposals/bits-and-bytes.adoc)

schorg, 2020-03-06:
Hex literals now possible. Feature branch is integrated into master.

closed 2020-03-06 by schorg