boschresearch/blech #12: support for instantaneous loops within activities

mterber, 2020-02-01:
**Is your feature request related to a problem? Please describe.**
Sometimes it is required to process arrays in the context of a running activity, e.g. for a copy operation like in the following example:
```
activity RF_sendCmdWord (dataTx: [...]uint8)(...) returns int8
    var tx_bytes: [SPI_DATA_SIZE]uint8 // SPI transmit buffer.
    var rx_bytes: [SPI_DATA_SIZE]uint8 // SPI receive buffer.
    var tx_len: uint16 = 0 // Number of bytes to send.
    ...
    // Send the provided tx data.
    memcpy2spi(dataTx, 0, tx_len, dataLen)(tx_bytes)
    res = run SPI_txRxBytes(tx_bytes, tx_len)(rx_bytes)
    ...
end
```
Here, _dataTx_ shall be copied to _tx_bytes_. This requires to loop over the array bytes within a single reaction. However, this is not allowed in activities, only in functions. So whenever I have to loop over arrays in an activity I have to create special function for that purpose.

**Describe the solution you'd like**
I would like to be able to execute instantaneous loops within activities directly.

frameworklabs, 2020-03-26:
I think this request would confuse more than it helps. I like the simple semantics currently in place. Creating instantaneous helper functions as you did with `memcpy` seem to me to be the better approach.

mterber, 2020-03-28:
Maybe a compromise could be to support variable array lengths. This way, I could at least reuse the same (generic) memcpy() function for different activities that deal with arrays of different size. Otherwise you end up with an impressive line-up of helper functions that, in principle, do exactly the same thing.

frameworklabs, 2020-03-30:
Maybe some limited generic support like below would help here some day:
```
function memcpy<L, T>(from: [L]T)(to: [L]T)
    var i: nat32
    while i < L repeat
        to[i] = from[i]
    end
end
```

mterber, 2020-04-01:
That sounds good from a users point of view. +1:

closed 2024-09-27 by NikolaBabadzhanov