boschresearch/blech #9: support for strings

mterber, 2020-02-01:
**Is your feature request related to a problem? Please describe.**
Today, Blech does not support string handling. Means whenever I want to print / send data in a human readable format, e.g. for showing it on a user interface (e.g. display) or for logging or debugging purpose, I have to pass the data to the C environment. In the C env I have to implement a special "converter function" for each and every individual string format that I need in my application. This is a tedious and error-prone task.

**Describe the solution you'd like**
I would like to be able to deal with strings directly in Blech. So for example to convert my binary data into a string, similar to printf, sprintf, etc. in C. Example:

```
var i: uint8 = 42
var i_as_string: str = string.format("The value of i is: %u", i)
```
Maybe it is possible to take advantage of the corresponding C functions like sprintf() for a similar implementation in Blech!?

schorg, 2020-08-01:
In branch "feature/string-literals" I started string handling in Blech.
For the next step you can at least created debug output with external C-functions
```
var i: uint8 = 42

@[CFunction (binding = """
                       printf("The value of i is: %u", $1)
                       """, 
             header = "string.h")]
extern function debugPrintNat8(i: nat8)

... var i: nat8 = 42
    debugPrintNat8(i)
```
The binding will become the right side of a macro.
For this purpose we allow multi-line and verbatim (raw) strings.
Later on these will also be used for string values in Blech.

schorg, 2020-08-07:
For the time being raw strings have been postponed. See feature/string-literals (https://github.com/boschresearch/blech/commit/6ae108ec1c4d99485bf17c88d62266480d8e38d7)

closed 2024-09-27 by NikolaBabadzhanov