boschresearch/blech #14: Syntax error for "next" as identifier

mterber, 2020-03-28:
**Describe the bug**
When I use the word "next" as an identifier for a variable I receive a syntax error. If I rename the variable, e.g. to "next1", then the error is gone.

> syntax error 'next'.parsing
test.blc(9, 1): start of chunk.
test.blc(10, 9): unexpected token.

**To Reproduce**
```
function cbuf_push () (head: nat16)
    var next: nat16 = head + 1
end
```

**Expected behaviour**
I would expect no syntax error because according to https://github.com/boschresearch/blech-doc/blob/master/documentation/blech-user-manual/lexical.adoc "next" does not seem to be a keyword (or special in any sense).

**Screenshots**
[image]

schorg, 2020-03-28:
'next' is a reserved keyword. It's dual to 'prev' and will become necessary for parallel blech.

mterber, 2020-03-28:
Okay, thanks for the clarification.

closed 2020-03-28 by mterber