boschresearch/blech #24: Allow variable declaration in run result

schorg, 2020-05-06:
I would like to write this:
```
let summary = run Controller()
```
But the compiler wants me to separate this like that:
```
var summary: Summary 
summary = run Controller()
```
Any chance to align this with function invocations?

This feature request was originally proposed by @frameworklabs

schorg, 2020-05-06:
`run` is a statement and and cannot occur in a right-hand-side expression.

To emphasize the statement character we propose to use
```
run let summary = Controller()
```
This will also break the old syntax, which changes to:
```
var summary: Summary
run summary = Controller()
```

This proposal is the result of a discussion on slack: https://blech-lang.slack.com/archives/CE0K9FFB6/p1588692639029400

schorg, 2020-05-06:
Some remarks:
We would allow:
```
run let s = Ctrl()
run var s = Ctrl()
```
These are fresh memory locations which act as local variables. They also occur in the following statements
```
for let i: nat8 = 1, 10 do ... end
if let payload = anEvent then ... end // event presence test and unpacking
emit let evt = payload //  one-shot event
emit let evt: event // one-shot event without payload
```
Everywhere `var` is also possible.

We do **not** allow the combination of `run` and `emit` statement.
```
run emit let evt = CtrlReturnsPayload()
```
In this rare case you need two statements
```
run let payload = CtrlReturnsPayload()
emit let evt = payload 
```
For the fresh location we do **not** allow extern locations and references.
```
extern var, extern let
let ref, var ref
extern let ref, extern var ref
```
All of this will be encoded into the grammar.

schorg, 2020-05-25:
One more thing: Blech should allow to directly return with the result of a sub-activity.
To combine the `run` and `return` statement, we will write:
```
return run Ctrl()
```
Instead of
```
run let summary = Ctrl()
return summary
```
In puts the emphasis on `return` and not the other way round 
`run return Ctrl()`.

FriedrichGretz, 2020-07-29:
I believe both the in-place declaration of the receiver as well as the return of a run result are completed on the develop branch.
Provided we [document ](https://www.blech-lang.org/docs/user-manual/controlflow/statements/#run) this feature properly, it can be released into master.

FriedrichGretz, 2020-09-01:
Fixed with Release 0.6.0

closed 2020-09-01 by FriedrichGretz