boschresearch/blech #25: Allow assignment to a struct with let qualified fields as a whole

schorg, 2020-05-06:
Currently, in Blech you cannot assign a compound value to a struct with let qualified fields, after the initialisation
```
struct S 
    var a: int32 
    let: b: int32 
end
...
var x: S = {a=1, b=2}
var y: S = {} // all fields 0

y = x // <-- error not allowed because b is let-qualified.
y.a = ...  // this is allowed
```
Swift allows this. The reason is

> in the case of y = x you replace a compound value with a new one whereas in the case of y.a = ... you modify an existing value. 

This feature request was originally proposed by @frameworklabs.

The discussion on Slack is here: https://blech-lang.slack.com/archives/CE0K9FFB6/p1588687479012200

 Proposal: Adopt the Swift semantics, because it is more natural.

FriedrichGretz, 2020-12-09:
One more argument supporting semantics à la Swift:

With opaque types we do not know anymore if there are fields in the type and whether they are protected.

```
import M "anotherModule" // has opaque type T 

// inside some activity
    var x: M.T = M.init() // always ok
    x = M.init() // ok with proposed semantics but an undetectable issue in the old semantics
```

I therefore propose to change the type checker now accordingly.

FriedrichGretz, 2021-04-30:
Has been implemented a while ago and is explicitly checked in test/blechc/programs/newStructReinit2.blc since commit afdb9b5 
The documentation issue is in https://github.com/boschresearch/blech-doc/issues/7

closed 2021-04-30 by FriedrichGretz