boschresearch/blech #26: Compilation error with new nat data types

mterber, 2020-06-03:
**Describe the bug**
I recently updated an older Blech code to the newest syntax of blechc. The old code looked like this
```
const MSEC_PER_SYSTICK: uint8 = 10

// all in milliseconds
const DCF77_SYNC_LEN: uint16 = 1200 / MSEC_PER_SYSTICK
const DCF77_LOW_MIN: uint8 = 80 / MSEC_PER_SYSTICK
const DCF77_LOW_MAX: uint8 = 120 / MSEC_PER_SYSTICK
const DCF77_HIGH_MIN: uint8 = 180 / MSEC_PER_SYSTICK
const DCF77_HIGH_MAX: uint8 = 220 / MSEC_PER_SYSTICK
```
and it compiled without any errors or warnings. Then I introduced the new data types and received the following error:
[image]

For me, this error message is not plausible. I explicitly declare DCF77_SYNC_LEN as nat16. So why should the compiler try to represent it as a nat8? Is it because MSEC_PER_SYSTICK is declared as nat8? What is the difference compared to the old version of the code?

I would not expect a compilation error in this scenario.

Version info:
```
311c82f3d1bdfe587116167f78ea4131e6fb12a5
Blech Compiler 0.5.3+0  Copyright (C) 2019-2020 see blech-lang.org
```

FriedrichGretz, 2020-06-03:
Yes, the reason of the error message is because MSEC_PER_SYSTICK is nat8.

The type checker tries to infer what kind of division is needed based on the two operands (regardless of what the left hand side of the assignment may be). The type checker could promote MSEC_PER_SYSTICK to nat16 to fit the larger operand 1200 but in many other cases this could be a mistake. 
Only number literals are promoted in the sense that "nat16_number + 1" is okay, because (unlike a named variable) the literal "1" is not declared to be exactly of type nat8 and thus can be used everywhere it fits. 

Try annotating ": nat16" to MSEC_PER_SYSTICK or use "as" to fix this issue.

mterber, 2020-06-03:
Okay, got it – thanks!

Does that mean that the behavior of the Blech compiler changed with respect to this? Because my older code with the previous version of the data types has been built without any error messages.

FriedrichGretz, 2020-06-03:
Yes, I believe the old compiler simply checked if a common size for the operands exists. That was before we introduced type annotations.

FriedrichGretz, 2020-06-03:
@schorg before this issue is closed, we need to concretise the documentation of https://www.blech-lang.org/docs/user-manual/expressions/#binary-arithmetic-operations (and the following examples)

closed 2024-09-27 by NikolaBabadzhanov