boschresearch/blech #44: Singleton function should not be callable in static initialiser

schorg, 2021-01-13:
**Describe the bug**
Singleton functions are functions that have a side effect on global resources.
Calling them in static initialiser should not be allowed, because the initialiser might change its value in every initialisation.

**To Reproduce**
```blech
@[CFunction (binding ="c_f", header = "c_f.h")]
extern singleton function f () returns bool

struct S 
    var b: bool = f() // singleton in static initialiser -> non-determinism
end 

@[EntryPoint]
activity Main ()
    var s: S
    await s.b
end
```
Compiles without errors.

**Expected behaviour**
The type checker should detect that the call `f()` might have a side effect and flag an error.

FriedrichGretz, 2021-02-05:
In general it should be checked that initialisers in a static context are static -- compile time evaluated -- expressions.

Currently, it so happens that (non-singleton) functions may be used because they can only access read-only const/param data which does not change throughout the run time of the program and thus even though the initialisation is executed each time a new struct of this type is created, it will always produce the same value.

This may change in the future, if functions (like activities) can access external data. Then a function with a `@[CInput ...]` is not a singleton but may return different values each time. It then would become necessary to distinguish between compile-time functions and run-time functions.

For now, we need to decide if we only rule out singleton functions or rule out the use of functions in static initialisation context completely (until we introduce a compile-time subexpression language).

schorg, 2021-02-05:
If we allow (non-compiletime) functions, then the call is kind-of lazy. 

I would vote for compile-time expressions - which means for now, we rule out the use of functions completely.

FriedrichGretz, 2021-02-05:
Yes this is also my preference: rule out functions from expressions in static contexts.

commit 0eaf3f03 (2021-02-08): Ensure that in struct declarations the struct fields are initialised by static expressions. Fixes #44
closed 2021-02-08 by FriedrichGretz