boschresearch/blech #46: Handle abstract types in type checking

schorg, 2021-01-21:
**Describe the todo**
When compiling signatures, abstract types are not handled.
```blech
module exposes init

struct S 
    var x: bool
end

function init() returns S 
    return {x = true}
end
```
gets signature 
```blech
signature

@[ComplexType]
type S

function init() returns S
```
`type S` is currently not handled by the typechecker.
Attributes `@[ComplexType]` and `@[SimpleType]` are currently not handled by the attribute checker.

**Link to a branch**
Typechecking and attribute checking [codegeneration for feature/modules](/boschresearch/blech/tree/feature/modules)
[Testcase](https://github.com/boschresearch/blech/blob/feature/modules/test/TestPhases/importchecking/valid/abstractfromcomplex.blc)

**Task list**
To mark completed todos, use [x]

- [ ] Complete type checking for abstract types
- [ ] Complete attribute checking for abstract types
- [ ] Write some test cases

FriedrichGretz, 2021-02-23:
The type checker now does check annotations for opaque types and checks initialisation of opaque typed variables.

However, export inference is not checked. Can we ensure that, e.g.
 - an implicitly exported simple type is correctly identified as "simple opaque" by the inference checker

schorg, 2021-02-23:
Yes we can. 

The import compilation for export inference only runs the phases before the type checker and then generates the signature.

I added a valid test case to the export inference tests. Just run the test: exportinference/valid/useabstractsimple.blc and inspect the generated signature in: exportinference/imports/abstractsimple.blh

closed 2021-06-08 by schorg