boschresearch/blech #48: Detect missing import in type checker

FriedrichGretz, 2021-03-04:
**To Reproduce**
Minimal source code to reproduce the behaviour.
If applicable, compiler invocation and arguments.

**Expected behaviour**
Error message

schorg, 2021-03-04:
Example 3 Modules in an import chain:

```
// file service.blc
module exposes S

struct S
    var b: bool
end
---
// file client.blc
import s "service"

module exposes C

const C: s.S = { b = true }
---
// file main.blc
import c "client"
// import s "service"

module exposes Main
activity Main ()
    let flag = c.C
    // let flag2: s.S = c.C
    await true
end
``` 
`let flag = c.C` has a type the originates from `service`, but it does not have a name here.

After a closer inspection. I tend to not flag this as an error.
The program is not really wrong and does also not leak any hidden information.
As soon as you need a type declaration or access to function from `service`, the  `import s "service"` will be required - simply for name checking purposes.
In the C implementation the .h-file also include indirectly.
The language server still can refer to file `service`.

It's not perfect - but it does not break any semantic guarantees.
Actually there is not perfect solution: How should the import be named? 
Is `import _ "service"` correct, only do make the dependency explicit.

Proposal: We should live with the implicit dependency

FriedrichGretz, 2021-03-05:
Ok, let's document this in the manual and then close the issue.

closed 2024-09-27 by NikolaBabadzhanov