boschresearch/blech #56: Polluted keyword space

FriedrichGretz, 2021-05-04:
**Describe the bug**
Due to past concept evaluations there are some keywords in the implemented grammar of the Blech language which are not documented in https://www.blech-lang.org/docs/user-manual/lexical/
In particular, there are some which probably will never make it into the language in that particular syntax such as "up" "down" "count", ...

**To Reproduce**
The code shown in the blog article https://www.blech-lang.org/blog/2020/11/23/a-module-system-for-blech/#module-implementation fails to compile due to parser errors regarding the keyword "count" used as a field name.

**Expected behaviour**
Code shown on the website must compile.

We could fix the blog article and add a *complete* list of reserved but unused keywords to the documentation.

I'd rather favour to throw out grammar that is purely experimental. Thus making the blog's code correct without a change. And then add a list of reserved keywords to the documentation that we leave intentionally in the grammar for future work such as "ref", "signal", "assume", "assert", ...

schorg, 2021-05-05:
Removed clock definitions and their keywords from the grammar.
They certainly will return in a future release - maybe as annotations or something else.
For the time being - keep the number of keywords low.
Still todo: add a list of reserved keywords for future work to the documentation.

FriedrichGretz, 2021-06-23:
Updated documentation, cf. https://github.com/boschresearch/blech-doc/commit/93507fcad6cd64127d0da57fdbf1b4cf2a31ed0d

closed 2021-06-23 by FriedrichGretz