boschresearch/blech #61: Allow doc comments for modules

FriedrichGretz, 2021-05-21:
**Is your feature request related to a problem? Please describe.**
A doc comment `/** */` before a `module` declaration result in a syntax error.

**Describe the solution you'd like**
The user might want to describe the module and also see this description in the generated C file.

**Describe alternatives you've considered**
The comment may placed *under* the module declaration and is thereby attached to the first top level declaration of the module but this seems unnatural.

schorg, 2021-05-21:
Should be possible, but not for imports - due to nasty shift-reduce conflicts.

schorg, 2021-05-21:
Syntax changed. See commit 83591e462ef7ae01b3cf9fe3e0d3b79450511307.

Still todo: Store annotation in AST. Generate doc comment in signatures, .c and .h files.

schorg, 2021-05-23:
Doc comments for modules are now stored in AST and typed AST.
module (and signatures) only allow for doc comment annotations.
This is checked during annotation checking. 
See commit 919dc835be1a80cbdf155e0268ada33b9cd7e068
Still todo: Generation of doc comments in signatures, .c and .h files.

schorg, 2021-05-24:
Module doc comments are now generated to signature, .c-files and .h-files. See commit 63651fcaf4ebf34ed902f431166afdebed48c9da

closed 2021-05-26 by schorg